beagleboard/beaglev-starlight #10: Insufficient clearance on mounting holes

pdp7, 2021-04-28:
From @marckleinebudde in the [forum](http://forum.beagleboard.org/):
> the clearance on the raspi compatible mounting holes is not that great. Also having copper around the holes would be good, too.
[image]

mitag, 2021-04-28:
As a workaround, you could perhaps replace the M 2.5 spacers with M 2 spacers and place a nylon washer underneath.

FionaYu20180326, 2021-04-29:
> As a workaround, you could perhaps replace the M 2.5 spacers with M 2 spacers and place a nylon washer underneath.
Thanks for your feedback. This is a mounting hole compatible with raspberry pi. We will modify this design to move the chip.

marckleinebudde, 2021-04-29:
It even works with the metal spacer. If I'm careful the spacer doesn't touch any pins of the chip. Just make sure it's fixed in the next board revision.

closed 2021-07-30 by pdp7