beagleboard/beaglev-starlight #11: suggestion to improve debug header signal levels

pdp7, 2021-05-04:
@bbbsnowball commented:
> The “Recover the bootloader” section shows a normal 3.3V USB-TTL adapter connected to the debug header. The signals on that header are labelled as 1V8. I think this ok anyway because there is a level shifter for RX (built with a pullup and a diode). However, be sure to get the connections right because there is a higher chance than usual to damage the SoC by connecting TX to TX. In addition, it is possible (but unlikely) that the TX levels are too low for some 3.3V adapters. We might want to collect some feedback on whether the usual ones (CH340, Prolific, FTDI) are ok.
> 
> The debug header is not populated on my board. If this is normal, you might want to mention that in the guide.

refer to forum post:
https://forum.beagleboard.org/t/update-bootloader-ddr-init-boot-u-boot-and-recover-bootloader/29558/7
(let me [Drew] know if you don't have access)

BBBSnowball, 2021-05-04:
Thanks. Just to be clear about this: I don't consider this a bug. This kind of cheap level shifter is well suited for the UART.

I only suggest that we could add some information to the guide:
1. Make sure to get the connections right! And maybe some guidance on how to measure which pin is RX/TX. Unfortunately, some adapters label it as "this is TX" and others as "connect this pin to TX" so you can't trust the labels.
  - Most adapters probably won't drive enough current to actually damage the SoC.
  - The most obvious labels are RX and TX plus an arrow that shows the data direction. We could add those to the board silk screen but there is hardly any space for it - and we would need the same labelling on the UART adapter.
2. This should work with any 3.3V adapter but there is a slight chance that it doesn't. If so, try another one. The following are known to work: ...

Both issues are only in theory. We can very well ignore them until we have some evidence that they can happen in practice.

BBBSnowball, 2021-05-07:
There is a [report](https://forum.beagleboard.org/t/beaglev-starlight-beta-development-report/29641/12) that PL2303 is not working. [The datasheet](https://cdn-shop.adafruit.com/datasheets/PL2303HX.pdf) says that an input should be at least 70% of VCC to be seen as high, which is 2.3V. 70% is a very common value so I assume that it is not the actual limit. However, 1.8V seems to be too low.

I recommend that we document the issue and don't change the board because most users won't use the debug header and advanced users will likely have more than one type of UART adapter.

pdp7, 2021-05-07:
Thank you.  @lakshanthad could you add to the getting started?

BBBSnowball, 2021-05-07:
CP2102 is [reported to be fine](https://forum.beagleboard.org/t/update-bootloader-ddr-init-boot-u-boot-and-recover-bootloader/29558/12). [FTDI, as well](https://forum.beagleboard.org/t/update-bootloader-ddr-init-boot-u-boot-and-recover-bootloader/29558/15).

lakshanthad, 2021-05-10:
> Thank you. @lakshanthad could you add to the getting started?

Yes I will @pdp7

lakshanthad, 2021-05-10:
> CP2102 is [reported to be fine](https://forum.beagleboard.org/t/update-bootloader-ddr-init-boot-u-boot-and-recover-bootloader/29558/12). [FTDI, as well](https://forum.beagleboard.org/t/update-bootloader-ddr-init-boot-u-boot-and-recover-bootloader/29558/15).

https://beagleboard.slack.com/archives/C01QV90F5KM/p1620581877127100?thread_ts=1619136698.331300&cid=C01QV90F5KM

pdp7, 2021-05-19:
@FionaYu20180326  @lakshanthad concerns from [Akira in the forum](https://forum.beagleboard.org/t/soldering-debug-port-for-boatloader-recovery/29819/8):
> I just checked the schematics of the BeagleV and GPIO13/UART3_RX_1V8 (1.8v) from jh71x0 soc is directly connected to 40pin-Pi-Connector by pull-up register with 3.3v to convert it from 1.8v but the debug serial port has pull-up register with 1.8v which means it is expecting 1.8v logic level instead of 3.3v, so this will cause logic level mismatch and worse case, it will damage the jh71x0 SoC. I will try 1.8v logic level USB-serial converter. Connecting 3.3v usb-serial converters to 1.8v debut serial port is dangerous.

>  Is it possible to ask Seeed to revise the schematics on the product version of the BeagleV to use both 3.3v logic level for both 40-pin-Pi-Connector and debug serial port? Expecting the developers to buy both 3.3v and 1.8v usb-serial adapters are a bit high boundaries.

> And I am not so sure how the schottky diode (D6 PMEG3010CEJ) used on UART3_RXD_1V8 for the protection. It looks like the schottky diode (D6 PMEG3010CEJ) was meant to be in between UART3_TXD_1V8 and pin3 of J14. If I have a time this weekend I will trace with my scope.

mcd500, 2021-05-20:
@BBBSnowball taught me at the forum that the schottky diode (D6 PMEG3010CEJ) is for a level shifter to convert 3.3V TX signal from the  adapter to 1.8V SoC input. The forward voltage of the diode is only 330mV at 100 mA, it is a nice design keeping the BOM cost low.

mcd500, 2021-05-20:
@FionaYu20180326 @lakshanthad, As @BBBSnowball mentioned, as long as the developer does not connect the 3.3V TX output of the adapter to the RX input on the board, then it will not damage the board and it is ok.
I checked the datasheets of CP2102 and while FTDI and CP2102 expects 2.0V min to detect as high (Vih) and it is 1.5V for FTDI, the output of 1.8V level from  jh71x0 SoC to be detected as a high at adapter chips it is kind of pushing the limit of the datasheet of the usb-serial chips.
I prefer having a level shifter on both UART3_TXD_1V8 and UART3_RXD_1V8 to have more margins but it is up to the decision of the Seeed.

mcd500, 2021-05-20:
@FionaYu20180326 @lakshanthad @BBBSnowball I listed all four chips, cp2102, ftdi, ch340, pl2303 at the forum. https://forum.beagleboard.org/t/usb-to-serial-adapters-known-to-work/29798
I have two ftdi 3.3v adapters and both worked on a 1.8v debug serial converter as well as one of my cp2101.

closed 2021-07-30 by pdp7