beagleboard/beaglev-starlight #14: Enable TPS65086 restart interrupt

pdp7, 2021-05-04:
@esmil worked on TPS65086 restart in their [Linux 5.12 branch](https://github.com/esmil/linux/tree/starlight) and noticed that it would work better if the interrupt line from the TPS65086 wsa connect.

@esmil could you describe in more detail?

We will try to fix in the next PCB spin.

esmil, 2021-05-04:
The TPS65086 has an output pin IRQB (open-drain, needs pull-up) to signal an interrupts. This should be routed to a GPIO pin on the SoC that can be set up to generate an interrupt when the TPS65086 pulls the line low.

Note however, that 7V is needed on IRQB when programming the OTP memory on the TPS65086, so if this might damage the SoC and the TPS65085 is programmed in-circuit some solution must be found.

(Btw. thanks for the consideration, but him/his is fine. I'm just a regular boring straight white nerd :P)

pdp7, 2021-06-08:
@FionaYu20180326 This issue is about allowing the PMIC to signal Linux that the system is going to be reset.  @esmil noticed that TPS65086 has an output pin IRQB (open-drain, needs pull-up) to signal an interrupt. This should be routed to a GPIO pin on the SoC that can be set up to generate an interrupt when the TPS65086 pulls the line low.

closed 2021-07-30 by pdp7